KinesisCorporation/Adv360-Pro-ZMK #473: MacOS Ventura 13.2.1 BLE connection issues

phrfpeixoto, 2024-05-13:
Hello!

Let me start by stating that I've been using the latest firmware without any major issues with my M2 MacBook Pro running Sonoma.

Now the issue I'm trying to solve is connecting that same keyboard to my desktop Mac, running Ventura (13.2.1): It won't connect.
I switched the bt profile on the keyboard to 2 (blue LED), and they keyboard was discovered as expected.
When trying to connect, the spinner goes on for quite a while, and then stops. The "connect" button changes label to "disconnect", but the keyboard is still showing on the "found devices" list, and never moved the the list of "paired devices"

Trying to debug this I little further, I've collected some debug logs for the bluetooth subsystem:
```
sudo log stream --debug --predicate 'subsystem=="com.apple.bluetooth" and category!="Server.LE.Scan" and category!="CBStackBLEScanner"' | grep "871C0F8E-B5EF-447F-FF99-B8029D1C2A93" | grep -v "Already tracking address"
```

This command is grep'ed for readability. The log is available on [this pastebin](https://pastebin.com/4KPCUXkm)

Any help is appreciated.

ReFil, 2024-05-13:
Have you ensured the second profile is cleared? Sometimes there is residual pairing data in the second profile and a clear is required, even if it shows as open. On the default windows keymap it's mod+win

phrfpeixoto, 2024-05-13:
@ReFil I had never used any of the other profiles, but since you mentioned it, here's what I did:
- Executed the clear profile key combo on profile 2 (blue)
- Successfully paired it with my iPhone
- Removed is from my iPhone
- Executed the clear profile key combo again on profile 2 (blue). Led is again blinking rapidly
- Tried to pair again with my desktop, to no avail 😢

ReFil, 2024-05-13:
If it's successfully pairing on other devices then it seems like there's something up with your mac, what model is it?

phrfpeixoto, 2024-05-13:
@ReFil It's a custom-built intel-based PC running MacOS Ventura. I've been using this computer for 5+ years without Bluetooth issues. I'm currently paired with a Logitech MX Master mouse and ad-hoc usage of Sony headphones, also over Bluetooth...

It seems this could be some corner case incompatibility between the controller/firmware on the keyboard and the controller/firmware on my BT adapter. Hence why I came to Github...

ReFil, 2024-05-13:
What bluetooth adapter are you using? There was a known bug with intel AX2xx adapters but that's been addressed. Is your mouse the original MX master or one of the newer versions, The 3 uses BLE but i'm not sure about the older versions

phrfpeixoto, 2024-06-08:
I'm sorry! I missed some of your questions.
My mouse is a Logi MX Master 3S, pretty new I guess.

ReFil, 2024-06-14:
Is your firmware fully up to date? if not please resync your repo with upstream and update your board

ReFil, 2024-11-27:
Hi, just following up on this issue, is it still a problem?

phrfpeixoto, 2024-11-28:
I have not tried lately. Will try again today after work is done. Thanks!