KinesisCorporation/Adv360-Pro-ZMK #483: On MacOS the keyboard disconnects and re-connects frequently with bluetooth

clauxx, 2024-05-27:
On MacOS (14.2), the keyboard is connected via bluetooth, but it keeps disconnecting (for ~5s the profile led is blinking) and reconnecting back every few minutes. What's frustrating is that once it disconnects, the key that was pressed at the time is stuck while disconnected. Anything I can do to make the connection more consistent?

Here's my [config](https://github.com/clauxx/Adv360-Pro-ZMK/tree/V3.0)

trevor-scheer, 2024-06-13:
This seems to be resolved for me after updating my fork. Thank you very much! I think my connection was dropping 50-100 times a day for ~5-10 seconds, it was borderline unusable. No drops today.

KinesisCorporation, 2024-06-13:
Great! Let us know if the drops come back.

closed 2024-06-13 by KinesisCorporation