KinesisCorporation/Adv360-Pro-ZMK #487: No Space Left on drive error on When updating firmware for Advantage Pro 360

numenbit, 2024-05-29:
I was able to change my keybindings for the keyboard a few times did the firmware upgrade did left and right no problem.   I was getting some errors but I just ignored and everything updated with the new keybindings no problem.   

Now a few months later and I want to change my keybindings again and now I am trying to drag the file to the keyboard mounted drive on my Mac and I am getting not enough space on drive error.    

Because of this error I can not update my changes.   I tried deleting the files on the mounted volume and they always come back so not sure how to get more space.   Maybe my other updates filled up the keyboard drive? 

How can I make space I tried deleted one of the files that was around 1.4MB but that one kept coming back latest.u2 something like that cant remember.    Going to try to use the console but since keyboard is not enabled I guess I will have to attach another keyboard.   

Any ideas?

ReFil, 2024-05-29:
The drive is a virtual drive, so you shouldn't have a problem with free space. Deleting the files that are there wont change that. The drive should say it's 32MB approx, is that the case?

ReFil, 2024-06-20:
Following up on this, are you still having this issue?

closed 2024-11-27 by ReFil