KinesisCorporation/Adv360-Pro-ZMK #494: Macro not registering

Odaym, 2024-06-06:
Hi, great work on the keyboard, I really love it.

I'm on a late 2023 Mac running Sonoma. I have some shortcuts for switching to different workspaces:
[image]

I wanted to make these as macros so that I could have the Hotkeys (1 and 2 for example) run them in 1 button.

What I did was this:

- Add them to the list of Macros under Edit macros the in the GUI mapper
[image]

- Assign them to the desired keys in the GUI mapper
[image]

- Flash the new firmware successfully onto the keyboard

However, running them yields wrong/no results:

- Running the first hotkey (1), just produces an F1 action which displays mission control
- 2 does nothing

I tried to add an FN into the macro to see if it would help but it didn't, same result. The reason being is that on the Adv360 I do have to press FN to get to the actual F1/2/3 keys, but on my mac keyboard or the Typematrix, I just do Shift + F1.
[image]

This is my current config: https://github.com/Odaym/Adv360-Pro-ZMK/tree/V3.0

Any help would be appreciated, thanks a lot!

Odaym, 2024-06-06:
Thanks for the response, so Macros basically are keys in succession, there can be no holding down this key and pressing down that key meanwhile.

I opened an issue in the Keymap-editor repo but if you have any experience with it? would it look something like this? Basically a Macro in itself as well but it has Press/Tap options

[image]

Thanks!

ReFil, 2024-06-06:
Macros can have press, tap and release options. You'll want to use &macro_press LSHFT, &macro_tap F1, &macro_release LSHFT

Odaym, 2024-06-06:
ok clear, well I'll try this today and will write back what and how it works, but let me just keep this open till then, thanks