KinesisCorporation/Adv360-Pro-ZMK #501: How can I set USB mode to its own profile?

abreinig, 2024-06-18:
My use case is that I use the keyboard plugged into USB on one machine and then switch to other machines via Bluetooth. However, it appears when a USB cable is plugged in that connection is always active regardless of what Bluetooth profile I select. How can I set USB mode to a profile (like 5) so that I can switch between USB mode and Bluetooth mode without repeatedly unplugging the USB cable to the left side? For reference, this is how I use the glove80 keyboard which is also uses ZMK: https://github.com/moergo-sc/glove80-zmk-config . Thank you.

ReFil, 2024-06-19:
Use the &out behaviour documented here: https://zmk.dev/docs/behaviors/outputs

abreinig, 2024-06-21:
This works great, thank you!

closed 2024-06-21 by abreinig