KinesisCorporation/Adv360-Pro-ZMK #506: Phantom Keypresses

kelvinlauKL, 2024-06-23:
I've had this keyboard for over a year now. I've noticed that it occasionally does a "phantom" keypress while I'm typing. For example, when I try to type:

Hi, how are you doing?

It would sometime insert additional keystrokes:

Hi, hoow are you doing?

Is this a problem other folks have noticed as well?

kharoutinian, 2024-06-23:
I see similar behavior with my 360 pro a couple times a day.  For me it will almost always manifest with the "phantom" keypress after the next letter like this:

> Hi, howo are you doing?

I also see it with both of my 1st gen Advantage boards so I've been assuming its because of the way that I type. I'll start taking note of what letters I type when it happens again to try to make some correlations.

ReFil, 2024-06-23:
If this is presenting an issue you can try adjusting your debouncing settings https://zmk.dev/docs/features/debouncing

linhub15, 2024-07-09:
I ran into this issue as well, and documented some mitigation steps in this reddit post.

https://www.reddit.com/r/kinesisadvantage/comments/1buz2vv/looking_for_suggestions_duplicate_and_double_key/

In the end, the keyboard was replaced. Now, on the replacement keyboard it is happening again.

e.g. when I typed "replacement", it typed "replaccement", sometimes there are three letter occurrences as well.

sbg-dev, 2024-08-01:
I was just looking if a new FW is available because I'm facing the same issue that was already described and I found this issue. I'll try the suggested solution with the debouncing settings.,
Btw. this issue happened twice while writing this post. ☹️

KinesisCorporation, 2024-08-01:
Thanks for raising this issue. Our tech support team has talked with some of you and we've been researching this issue ourselves.

If people are experiencing this type of issue with completely different keyboards, this sounds like a garden-variety typo or perhaps a PC problem...

If people are seeing this issue exclusively on the 360 Pro then it could be a firmware/Bluetooth/settings issue (keyboard is sending extra strokes or debounce isn't optimized for your typing speed/style) or a hardware issue with a switch. If it was a switch defect, we would expect to see it on the same key over and over. If its a firmware problem we would expect to see it happen with all keys to some degree.

Please let us know what each of you are seeing and whether its specific to Bluetooth or USB operating modes so we can investigate further. Thanks!

kelvinlauKL, 2024-08-05:
> Thanks for raising this issue. Our tech support team has talked with some of you and we've been researching this issue ourselves.
>
> If people are experiencing this type of issue with completely [...]

I'm on bluetooth, advantage 360 pro. I don't think it's limited to one key - there doesn't seem to be rhyme or reason to when it duplicates keys.

It happens on both my Mac Studio and MacBook Pro, so I don't think it's computer specific. It's happened to me enough that I don't think it's a typo.

KinesisCorporation, 2024-08-06:
> I'm on bluetooth, advantage 360 pro. I don't think it's limited to one key - there doesn't seem to be rhyme or reason to when it duplicates keys.
>
> It happens on both my Mac Studio and MacBook Pro, [...]

Gotcha. So its random and not reproducible in anyway and not limited to one key.

- Would be good to switch to USB for a little bit and see if the issue goes away?

- Have you tried increasing your debounce settings from the default 5ms closer to 10-12ms?

https://zmk.dev/docs/features/debouncing

kbrah, 2024-08-27:
I face the same issue. For me it happens only with Bluetooth. When I'm plugged in with USB there are no phantom keypresses.

KinesisCorporation, 2024-08-28:
> I face the same issue. For me it happens only with Bluetooth. When I'm plugged in with USB there are no phantom keypresses.

Please try adjusting your debounce settings to 10-12ms and report back.

ebzzry, 2024-09-02:
I have done all the suggestions but I still have phantom keypresses.

KinesisCorporation, 2024-09-03:
Sorry to hear that but thank you very much for trying! 

Please submit a trouble ticket. We'd like to have you try a bluetooth adapter to see if that makes a difference. If it turns out its a keyboard-specific problem don't worry, you are covered under the original warranty.

https://kinesis-ergo.com/support/contact-a-technician/

ebzzry, 2024-09-08:
After trying to fix the problem for at least a year—applying patches, changing settings—I give up.

I switched to wired mode and couldn't be happier.

michamcr, 2024-09-14:
I am experiencing this issue with Kinesis Freestyle 2 that I just bought on Amazon - filed a ticket. At first I thought it was my PC, I switched to a different machine and still seeing the issue. I'll go try debouncing and see how I get on. I have a second Kinesis Freestyle 2 for years and that is rock solid, this brand new one, something strange going on, and it shows up mostly when I press "space". 😵

KinesisCorporation, 2024-09-15:
> I am experiencing this issue with Kinesis Freestyle 2 that I just bought on Amazon - filed a ticket. At first I thought it was my PC, I switched to a different machine and still seeing the issue. [...]

Hi Michael, Sorry you are having an issue with your Freestyle2 but thank you for submitting a ticket. We will work with you directly. This thread (and whole forum) is specifically for Advantage360. The Freestyle2 has entirely different electronic architecture so while the symptoms may be similar, there problems are not the same so I'm going to hide these comments. Have great day!.

Fomiller, 2024-09-24:
I am also seeing the describe issue with an additional problem of input delay, typing a word and having the characters not appear or waiting 1-2 seconds to have the characters appear. I am using an Advantage 360 pro on MacOS 14.4 and 14.6 using bluetooth. When switching to usb the issue seems to resolve itself

KinesisCorporation, 2024-09-24:
Hi @Fomiller. Sorry to hear you are having this problem. Please send us a support ticket using the link below. We'd like to get you a complimentary USB-A Bluetooth dongle to test whether that addresses your issue.

https://kinesis-ergo.com/support/contact-a-technician/

vincic, 2024-10-12:
I have two sets, one from this year which is giving me these problems, mainly with `r`.  Many times it is not registered at all and other times multiple. The issue is on both cable and bluetooth on MacOS and on cable with Linux. I tried the debouncing suggestion 3 weeks ago, but no difference.

Tried to downgrade to my v2 version which is running flawless on my other keyboard but the problem still persists. 

Just submitted a support ticket on the link suggested above.

KinesisCorporation, 2024-10-14:
This sounds more like a faulty switch.

strotter, 2024-12-06:
Out of curiosity is the keyboard selection for all Advantage keyboards? I guess a followup would be does the Advantage 360 differ from flashing any differently than the Advantage 22? I have 3 Advantage boards and they all started glitching identically on the exact same day. Every time I think I find a culprit key or circuit I clean it more heavily, but no good.

Why would all 3 of mine start doing this at the same time if it's hardware related? System update combined with the keyboard BIOS maybe?

KinesisCorporation, 2024-12-06:
@strotter Shane, this thread is exclusively for Advantage360 Pro. It sounds like you are having difficulties with your Advantage2 which has a totally different electronic architecture (and switches). Can you please confirm which keyboard you have?

If you have an Advantage2 we will delete your comment and invite you to submit a ticket so we can help with troubleshooting.

strotter, 2024-12-07:
I deleted it, however, can you point me to a specific thread that discusssed the exxact shame  phantom keypressses, the eseentence I just typed out for example.  I happens on delays from 10 through 3000ms. Any gidance is appreciate.

KinesisCorporation, 2024-12-07:
@strotter I'm still confused. Which model Kinesis keyboard are we talking about? There are no threads on Github regarding the Advantage2 (to my knowledge)

If the same thing happened to all 3 keyboards on the same day, that is suggestive that something changed with your PC, not the keyboard. Have you tried connecting them to a different PC? Please direct future correspondance to our tech support team so we can keep this thread focused on Advantage360 issues.

rmagatti, 2024-12-24:
I found this thread today. I'm experiencing key chatter on my _second_ Kinesis Advantage 360 pro. Second because I returned the first one due to key chatter issues, the second one was working fine for the first couple of months but to my dismay started giving me chatter issues on a couple of letters, removed the keycaps, cleaned it and chccarged it (there's the chatter on trying to type "charged") and most other keys are now seemingly fine except for "C".

No matter how much I clean it there's frequent chatter on "C".

KinesisCorporation, 2024-12-30:
Hey @rmagatti, please send us a support ticket and we can get you some help.

shiqicao, 2024-12-31:
I am having the same issue, is it possible the problem is from the switch?

KinesisCorporation, 2025-01-02:
@shiqicao please send us a support ticket and we can try to troubleshoot whether its a defective individual switch or something else going on.

https://kinesis-ergo.com/support/contact-a-technician/

kbrah, 2025-01-23:
This issue seems to be fixed for me with the Clique update #630. Haven't had any phantom key presses since installing it.

closed 2025-02-06 by ReFil