KinesisCorporation/Adv360-Pro-ZMK #510: Right side intermittently not pairing with left side. What is the cause?

abreinig, 2024-06-25:
What is/are the cause(s) of the right side not pairing with the left side? I have used a different ZMK board for months (glove80) and I turn both sides on and off many times every day and never had a single issue despite never paying attention to the order of turning the sides on or anything. It is infuriating the dance I have to do to get the sides paired on the kinesis advantage 360. Is it a code problem, a hardware problem, or what?

ReFil, 2024-06-25:
Are you running the latest firmware? Older versions had a known bug causing issues with split connections

abreinig, 2024-06-25:
I just came back to put my firmware version, and you beat me! Firmware version v3.0 , my branch is synced with upstream KinesisCorporation/Adv360-Pro-ZMK:V3.0

yodahuang, 2024-07-09:
I'm also having the same issue with my branch, synced with V3.0.

KyleCotton, 2024-07-13:
I'm also having this issue. It seems that plugging in the left module then disconnecting seems to rectify the issue - temporarily. 

I have just updated to the latest firmware but the issue seems to persist.

abreinig, 2024-07-17:
Can we please get an official response on this? Other ZMK boards do not have this problem. If it's a software problem, can you contract someone to fix it? If it's a hardware problem, are you aware of how to fix it? Will it be fixed in an upcoming hardware revision?

KinesisCorporation, 2024-07-17:
Hi All-

Sorry to hear you are having "syncing" troubles. We've talked with our team here are not seeing any widespread issues with syncing. Did your keyboards stop syncing overnight or did it happen after a firmware flash or other operation?

Why don't you try performing BOTH a 2.0 and 3.0 Settings Reset and then re-flashing your 3.0 firmware using the Reset files and instructions linked below. If that still doesn't resolve the issue, please submit a ticket and reference this thread and we'll escalate to a repair/replacement unit.

**Files:** 
https://kinesis-ergo.com/download/adv360-pro-settings-reset-version-2-0/
https://kinesis-ergo.com/download/adv360-pro-settings-reset-version-3-0/

**Instructions:**
https://kinesis-ergo.com/wp-content/uploads/Advantage360-Professional-Firmware-Update-Instructions-11.22.23-KB360-PRO-GBR.pdf
https://kinesis-ergo.com/wp-content/uploads/Advantage360-Professional-Settings-Reset-Instructions-11.22.23-KB360-PRO-GBR.pdf

**Ticket:** https://kinesis-ergo.com/support/contact-a-technician/

abreinig, 2024-07-17:
This should be reproducible. I have had same exact issue on 2 different new boards with latest firmware and resets. Steps to reproduce:
1. Turn both halves off.
2. Turn on right side first.
3. Turn on left side.

The right side will not pair to the left. Even if I turn it off and on it will not pair. It will only pair if I turn both sides off, and turn the left side on and wait, then turn the right side on. So, when I am turning both sides on, if I am not careful enough to turn on the left side first, they don't pair and I have to start over. Please repeat these steps and let me know if you have the same issue. Like I said, my other ZMK boards do not have this problem.

KinesisCorporation, 2024-07-17:
Thanks for the additional information.

Early versions of firmware required that the left side be powered on first but newer versions are not picky. We just flashed factory firmware and powered on the right first with no issue.

Does it always sync properly when the left side is powered on first? Does it stay synced in that situation even after the keyboard goes to sleep?

It would be worth performing the double 2.0 and 3.0 reset and then installing see if you can restore the keyboard to expected functionality. If that doesn't work, we'll pay to ship it back to Kinesis to troubleshoot and repair/replace as needed per your warranty.

abreinig, 2024-07-17:
Thanks for the response. The halves have always synced properly as long as I turn on the left side and wait a second before turning on the right side. They stay synced even after the keyboard goes to sleep. I will try the double 2.0 and 3.0 reset and report back my testing results.

KinesisCorporation, 2024-07-17:
That's good to hear as its always been the best practice for powering on the devices.

abreinig, 2024-07-17:
Steps performed:
1. Put left half into bootloader mode and flashed 2.0 settings reset. Success, power off
2. Put right half into bootloader mode and flashed 2.0 settings reset. Success, power off
3. Put left half into bootloader mode and flashed 3.0 settings reset. Success, power off
4. Put right half into bootloader mode and flashed 3.0 settings reset. Success, power off
5. Power on and plug in left half and enter bootloader mode, copy my modified 3.0 firmware file. Success, left powered on.
6. Power on the right half and enter bootloader mode, copy my modified 3.0 right firmware file. Success, both halves sync to each other.

Result: Same problem as before. Even if I turn on the right half a millisecond (not scientific) before the left half, they don't sync.

KinesisCorporation, 2024-07-17:
Thanks for trying that. The keyboard certainly "should" allow you to turn on the right side first so its weird that it does not. Since the its working perfectly with the left side being powered on first its unlikely a hardware issue.

Are the other ZMK split keyboards you referenced in the vicinity of this 360 and powered on? Perhaps there is some interference.

If you want to proceed with a repair/replacement please submit a ticket.

abreinig, 2024-07-17:
That's a good question about interference. I took the halves to different parts of my home away from other devices in an attempt to get away from something that could be interfering but the behavior didn't change. It's not the most scientific test, I could take them in my car and drive to a field for a better test. It does give me hope that you can't replicate this behavior.

bartlomiej-mucha-bpms, 2024-07-29:
I have similar problem. Here is a branch with my config, it's on top V3 branch https://github.com/bartlomiejmucha/Adv360-Pro-ZMK/commits/my-customisation-v3/
In my config I disabled bluetooth with (CONFIG_BT=n) as I use usb cable only.

To make keyboard work I have to unplug both usb cables, and then plug the left one first, and then after a second, plug the right one. If I do it too quickly, I have to repeat the process.

The keyboard is connected with original usb cables to the caldigit elements dock, and dock is connected to MacBook Air.

bartlomiejmucha, 2024-07-29:
I have similar problem. Here is a branch with my config, it's on top V3 branch https://github.com/bartlomiejmucha/Adv360-Pro-ZMK/commits/my-customisation-v3/
In my config I disabled bluetooth with (CONFIG_BT=n) as I use usb cable only.

To make keyboard work I have to unplug both usb cables, and then plug the left one first, and then after a second, plug the right one. If I do it too quickly, I have to repeat the process.

The keyboard is connected with original usb cables to the caldigit elements dock, and dock is connected to MacBook Air.

AntoniMarcinek, 2024-08-16:
I have a similar problem when using the [upgradekeyboards single USB cable for both sides](https://upgradekeyboards.com/products/advantage360-pro-charge-use-single-double-sleeved-usb-c-cable-upgrade): when I have USB-C ends connected to the keyboard's halves and connect the USB-A to PC or when all connectors are connected and I power on the PC, I got red blinking LEDs on the right side and neither half works. I also get a message from my Linux kernel when such problem happens:
```
usb 1-5: device descriptor read/64, error -110
```
The only way to recover is to take out both USB-Cs and put them in again (sequence doesn't matter as expected from the v3 firmware).

kbrah, 2024-08-27:
I'm also facing the same issue, and have had it for ages. It's especially annoying because when I boot my computer and the keyboard is plugged in with usb, it doesn't pair properly before I replug both halves in the left to right order. 

Could it be related to the OS/hardware of the computer? I'm currently running Asahi Linux on MacBook Pro and I think the issue was not present atleast in the booting case on my previous setup which was Arch Linux on a Thinkpad.

AntoniMarcinek, 2024-09-12:
I realised that the problem is not limited to the USB connection: if I simultaneously switch on both batteries when the USB cables are removed, then I get the same pairing problem. Some race condition? @KinesisCorporation @ReFil any comments on this?

ReFil, 2025-01-17:
A new update is now available for testing that should include some fixes to help with this bug. precompiled UF2 files are available here: https://kinesis-ergo.com/download/advantage360-pro-clique-beta-01-16-25/ for users without significantly complex custom keymaps, users with complex keymaps can build off the PR #630 whilst this firmware remains in the beta stage

kbrah, 2025-01-21:
So far it seems that this issue is gone with the clique update. Thank you!

abreinig, 2025-01-21:
Is this officially confirmed to be a software issue then?

ReFil, 2025-02-06:
@abreinig this should now be solved with the new update, if you could update your repo and report your experience that would be great, thank you!

ReFil, 2025-02-17:
Is anyone still encountering this issue? I will close it out in 24h otherwise as it seems to be solved with the latest update

guilhermetk, 2025-02-17:
I have not experienced this issue since upgrading to the latest firmware version a week ago. I keep both sides connected via USB to avoid charging them, so I had to unplug and plug them back in (left first) a few times a week before the update.

closed 2025-02-18 by ReFil