KinesisCorporation/Adv360-Pro-ZMK #511: USB HID protocol support (CONFIG_ZMK_USB_BOOT doesn't quite work)

edwintorok, 2024-06-26:
I am not able to use the keyboard to login on the macOS Sonoma 14.5 bootup screen on wired USB connection. I am able to use a Kinesis2 though.

I attempted to enable this with this commit: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/compare/V3.0...edwintorok:Adv360-Pro-ZMK:V3.0

That seems to work on macOS Sonoma 14.5: I can login now.

However it seems to completely break the keyboard on Linux: although the keyboard seems to work initially, it just stops typing after a while (e.g. I was not even able to type this entire line in that mode). Restoring to factory V3.0 makes it work again.

See https://zmk.dev/docs/config/system#usb
> CONFIG_ZMK_USB_BOOT	bool	Enable USB Boot protocol support	n
> By default USB Boot protocol support is disabled, however certain situations such as the input of Bitlocker pins or FileVault passwords may require it to be enabled

So I haven't opened a PR, but rather a bug: how can I get Kinesis Adv360Pro to work on the macOS login screen, without breaking Linux?

ReFil, 2024-07-01:
Strange, are there any relevant dmesg logs when this happens?

edwintorok, 2024-07-01:
See below for all the USB messages, unfortunately I didn't make a note of the timestamps when it was broken vs when it was working.

BTW I've enabled the setting in both the left and right configs, is that perhaps wrong and I should've only changed the left config?

```
Jun 27 00:08:16 fedora kernel: usb 5-2.4.2: USB disconnect, device number 5
Jun 27 00:11:13 fedora kernel: usb 5-2.4.3: USB disconnect, device number 6
Jun 27 00:11:13 fedora kernel: usb 5-2.4.3: new full-speed USB device number 8 using xhci_hcd
Jun 27 00:11:14 fedora kernel: usb 5-2.4.3: New USB device found, idVendor=239a, idProduct=00b3, bcdDevice= 1.00
Jun 27 00:11:14 fedora kernel: usb 5-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 27 00:11:14 fedora kernel: usb 5-2.4.3: Product: ADV360PRO
Jun 27 00:11:14 fedora kernel: usb 5-2.4.3: Manufacturer: Kinesis
Jun 27 00:11:14 fedora kernel: usb 5-2.4.3: SerialNumber: 216F8A72282AFAC7
Jun 27 00:11:14 fedora kernel: usb-storage 5-2.4.3:1.2: USB Mass Storage device detected
Jun 27 00:11:14 fedora kernel: scsi host6: usb-storage 5-2.4.3:1.2
Jun 27 00:11:14 fedora kernel: cdc_acm 5-2.4.3:1.0: ttyACM0: USB ACM device
Jun 27 00:11:14 fedora kernel: usbcore: registered new interface driver cdc_acm
Jun 27 00:11:14 fedora kernel: cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
Jun 27 00:11:15 fedora kernel: scsi host6: scsi scan: INQUIRY result too short (5), using 36
Jun 27 00:11:15 fedora kernel: scsi 6:0:0:0: Direct-Access     Adafruit nRF UF2          1.0  PQ: 0 ANSI: 2
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: Attached scsi generic sg3 type 0
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: [sdd] 65801 512-byte logical blocks: (33.7 MB/32.1 MiB)
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: [sdd] Write Protect is off
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: [sdd] Mode Sense: 03 00 00 00
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: [sdd] No Caching mode page found
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: [sdd] Assuming drive cache: write through
Jun 27 00:11:15 fedora kernel:  sdd:
Jun 27 00:11:15 fedora kernel: sd 6:0:0:0: [sdd] Attached SCSI removable disk
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: USB disconnect, device number 8
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 269 op 0x1:(WRITE) flags 0x0 phys_seg 5 prio class 2
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 269, lost async page write
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 270, lost async page write
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 271, lost async page write
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 272, lost async page write
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 273, lost async page write
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 11 op 0x1:(WRITE) flags 0x100000 phys_seg 1 prio class 0
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 11, lost async page write
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 517 op 0x1:(WRITE) flags 0x0 phys_seg 1 prio class 2
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 517, lost async page write
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 12 op 0x1:(WRITE) flags 0x100000 phys_seg 4 prio class 2
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 12, lost async page write
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 13, lost async page write
Jun 27 00:11:31 fedora kernel: Buffer I/O error on dev sdd, logical block 14, lost async page write
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 517 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
Jun 27 00:11:31 fedora kernel: FAT-fs (sdd): unable to read inode block for updating (i_pos 8275)
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 517 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Jun 27 00:11:31 fedora kernel: FAT-fs (sdd): unable to read inode block for updating (i_pos 8274)
Jun 27 00:11:31 fedora kernel: device offline error, dev sdd, sector 517 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 2
Jun 27 00:11:31 fedora kernel: FAT-fs (sdd): unable to read inode block for updating (i_pos 8279)
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: new full-speed USB device number 9 using xhci_hcd
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: New USB device found, idVendor=29ea, idProduct=0362, bcdDevice= 3.05
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: Product: Adv360 Pro
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: Manufacturer: Kinesis Corporation
Jun 27 00:11:31 fedora kernel: usb 5-2.4.3: SerialNumber: 216F8A72282AFAC7
Jun 27 00:11:31 fedora kernel: input: Kinesis Corporation Adv360 Pro Keyboard as /devices/pci0000:00/0000:00:08.1/0000:12:00.4/usb5/5-2/5-2.4/5-2.4.3/5-2.4.3:1.0/0003:29EA:0362.000A/input/input19
Jun 27 00:11:32 fedora kernel: hid-generic 0003:29EA:0362.000A: input,hidraw3: USB HID v1.11 Keyboard [Kinesis Corporation Adv360 Pro] on usb-0000:12:00.4-2.4.3/input0
Jun 27 00:11:32 fedora kernel: umount: attempt to access beyond end of device
Jun 27 00:11:32 fedora kernel: FAT-fs (sdd): unable to read boot sector to mark fs as dirty
[... cut ...]
Jun 27 00:21:13 fedora kernel: usb 5-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jun 27 00:21:13 fedora kernel: usb 5-2.4.3: Product: Adv360 Pro
Jun 27 00:21:13 fedora kernel: usb 5-2.4.3: Manufacturer: Kinesis Corporation
Jun 27 00:21:13 fedora kernel: usb 5-2.4.3: SerialNumber: 216F8A72282AFAC7
Jun 27 00:21:13 fedora kernel: input: Kinesis Corporation Adv360 Pro Keyboard as /devices/pci0000:00/0000:00:08.1/0000:12:00.4/usb5/5-2/5-2.4/5-2.4.3/5-2.4.3:1.0/0003:29EA:0362.0016/input/input31
```

ReFil, 2024-07-03:
It should only be enabled on the left side. Looking at that log it seems like the board is resetting

edwintorok, 2024-07-03:
So I tried enabling the config on just the left side: https://github.com/edwintorok/Adv360-Pro-ZMK/actions/runs/9785119127

Still the same problem as before, both halves work initially but then it disconnects like before after a short time, and no letters appear, regardless of whether I try typing on left or right half:
```
[  660.913490] usb 5-2.4.3: new full-speed USB device number 20 using xhci_hcd
[  661.026334] usb 5-2.4.3: New USB device found, idVendor=29ea, idProduct=0362, bcdDevice= 3.05
[  661.026339] usb 5-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  661.026341] usb 5-2.4.3: Product: Adv360 Pro
[  661.026342] usb 5-2.4.3: Manufacturer: Kinesis Corporation
[  661.026343] usb 5-2.4.3: SerialNumber: 216F8A72282AFAC7
[  661.089588] input: Kinesis Corporation Adv360 Pro Keyboard as /devices/pci0000:00/0000:00:08.1/0000:12:00.4/usb5/5-2/5-2.4/5-2.4.3/5-2.4.3:1.0/0003:29EA:0362.0008/input/input17
[  661.141760] hid-generic 0003:29EA:0362.0008: input,hidraw3: USB HID v1.11 Keyboard [Kinesis Corporation Adv360 Pro] on usb-0000:12:00.4-2.4.3/input0

[  689.402033] usb 5-2.4.3: USB disconnect, device number 20
```

Interestingly pressing Mod+(1) brings it back to life, I can flash factory firmware and it all works again:
```
[  718.770246] usb 5-2.4.2: new full-speed USB device number 21 using xhci_hcd
[  718.870243] usb 5-2.4.2: config 1 has no interfaces?
[  718.882247] usb 5-2.4.2: New USB device found, idVendor=29ea, idProduct=0362, bcdDevice= 3.05
[  718.882251] usb 5-2.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  718.882253] usb 5-2.4.2: Product: Adv360 Pro rt
[  718.882255] usb 5-2.4.2: Manufacturer: Kinesis Corporation
[  718.882257] usb 5-2.4.2: SerialNumber: 2C6E8D700BABAF50
[  899.249526] usb 5-2.4.3: new full-speed USB device number 22 using xhci_hcd
[  899.360948] usb 5-2.4.3: New USB device found, idVendor=239a, idProduct=00b3, bcdDevice= 1.00
[  899.360954] usb 5-2.4.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  899.360957] usb 5-2.4.3: Product: ADV360PRO
[  899.360958] usb 5-2.4.3: Manufacturer: Kinesis
[  899.360960] usb 5-2.4.3: SerialNumber: 216F8A72282AFAC7
[  899.423045] cdc_acm 5-2.4.3:1.0: ttyACM0: USB ACM device
[  899.425973] usb-storage 5-2.4.3:1.2: USB Mass Storage device detecte
```

I'm not sure how to debug this further, probably needs some debug logging enabled somewhere.
Can you reproduce this with the keyboard that you have and my config/firmware? Or can you suggest some additional config options that I can try to make it work, or to gather more debug information on what goes wrong?

edwintorok, 2024-07-03:
Ah there is another setting : https://github.com/zmkfirmware/zmk/pull/1793#issuecomment-1810058295
Looking at Kconfig enabling CONGIG_ZMK_USB_BOOT should've also enabled that one, but looking at the github action build log it doesn't seem to be enabled.

If I enable ZMK_USB_BOOT using menuconfig then the diff of the created config looks like this:
```diff
--- firmware/x0 2024-06-26 23:53:59.603521418 +0100
+++ firmware/xc 2024-06-26 23:52:48.516787372 +0100
@@ -34,7 +34,7 @@
 # Output Types
 #
 CONFIG_ZMK_USB=y
-# CONFIG_ZMK_USB_BOOT is not set
+CONFIG_ZMK_USB_BOOT=y
 CONFIG_USB_NUMOF_EP_WRITE_RETRIES=10
 CONFIG_USB_HID_POLL_INTERVAL_MS=1
 CONFIG_ZMK_BLE=y
@@ -2161,7 +2161,7 @@
 CONFIG_USB_MAX_NUM_TRANSFERS=4
 CONFIG_USB_REQUEST_BUFFER_SIZE=128
 CONFIG_USB_MAX_ALT_SETTING=8
-# CONFIG_USB_DEVICE_SOF is not set
+CONFIG_USB_DEVICE_SOF=y
 # CONFIG_USB_DEVICE_BOS is not set
 # CONFIG_USB_DEVICE_OS_DESC is not set
 CONFIG_USB_SELF_POWERED=y
@@ -2192,7 +2192,7 @@
 # CONFIG_ENABLE_HID_INT_OUT_EP is not set
 CONFIG_HID_INTERRUPT_EP_MPS=16
 CONFIG_USB_HID_REPORTS=1
-# CONFIG_USB_HID_BOOT_PROTOCOL is not set
+CONFIG_USB_HID_BOOT_PROTOCOL=y
 # CONFIG_USB_DEVICE_AUDIO is not set
 # CONFIG_USB_DEVICE_STACK_NEXT is not set
 # CONFIG_USB_HOST_STACK is not set
```

So I'll try a firmware with those 3 settings.

edwintorok, 2024-07-03:
Unfortunately that didn't work either (firmware at https://github.com/edwintorok/Adv360-Pro-ZMK/actions/runs/9785360171)

ReFil, 2024-07-03:
Maybe disabling nkro will help?

AntoniMarcinek, 2024-08-13:
Hello, is there any progress on this? I have the same problem. I didn't check with nkro disabled. I can revive the keyboard by switching to bluetooth and then reconnecting the cables.

AntoniMarcinek, 2024-08-16:
I disabled the NKRO and it doesn't help. I have the keyboard operational in GRUB and then it is not operational when the OS loads.

AntoniMarcinek, 2024-09-12:
@ReFil @KinesisCorporation Is there any progress on this?

AntoniMarcinek, 2025-02-11:
I also tested after #630 was merged and I confirm that now the keyboard works as expected. @ReFil thank you very much!

edwintorok, 2025-02-11:
Thanks, I can also confirm that enabling CONFIG_ZMK_USB_BOOT on both left/right seems to work now (previously when I tried that it didn't work reliably).
I can now use the keyboard in Linux, and also use it on the Mac OS login screen and after logging in on the Mac too.

I need to test this for a few more days, but the new firmware+this option appears to work great so far!

For reference here is the [build](https://github.com/edwintorok/Adv360-Pro-ZMK/actions/runs/13273944440) and [commit](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/f4b757f3dcb0d13e55032b6d1a12888057b7f514) I used, which is very similar to https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/511#issuecomment-2648848823

ReFil, 2025-02-17:
@edwintorok how has the additional testing gone? Is the issue fully solved for you?

edwintorok, 2025-02-17:
> @edwintorok how has the additional testing gone? Is the issue fully solved for you?

All good on both Linux and Mac.

closed 2025-02-17 by ReFil

edwintorok, 2025-02-17:
> Ok awesome, with that in mind i think i'll close this issue out as solved now. Thanks for getting back to me :)

Just to clarify it only works with the additional commit. I can try to open a PR.

ReFil, 2025-02-17:
It's expected it'll only work with that config option enabled, its left disabled by default for compatibility reasons

edwintorok, 2025-02-17:
> It's expected it'll only work with that config option enabled, its left disabled by default for compatibility reasons

Should there perhaps be 3 builds of the firmware? One without Clique, one with Clique, and one with Clique+boot proto?