KinesisCorporation/Adv360-Pro-ZMK #518: Build always fails

dmh43, 2024-07-12:
Can someone help me figure out why my build fails on this branch? https://github.com/dmh43/Adv360-Pro-ZMK/tree/dh-new

All I've done is update the keymap.

ReFil, 2024-07-12:
In your macros.dtsi file there are two macro_vrackets and macro_kinesis macros. Delete them and it should work

https://github.com/dmh43/Adv360-Pro-ZMK/blob/v3-local/config%2Fmacros.dtsi

dmh43, 2024-07-12:
It seems every time I make any change these days my build fails, is there some way to add validation to the GUI?

ReFil, 2024-07-12:
The additional macros are probably in the macros section of the GUI as that is stored in an additional file. You should be able to remove them from the macros gui