KinesisCorporation/Adv360-Pro-ZMK #519: Bricked keyboard / Left half will not mount

Joseph-Lion, 2024-07-12:
Yesterday the Adv360Pro randomly disconnected and when I opened the bluetooth settings of my computer to ‘connect’ the device it went into a very common loop of ‘connecting’ and then failing which calls for flashing and re-adding firmwarez

When I connect the left half of the Adv360Pro to my Macbook and either key press Mod+Bootloader or double press the physical reset button with the 3 LEDs and backlights flash, however, nothing mounts. When I repeat this step with the right half it immediately mounts and the LEDs turn green.

This is very concerning as it’s bricked the keyboard. Any advice on how to get this board working again would be wonderful, thank you

ReFil, 2024-07-12:
If the keyboard is not booting you need to use the double press reset method to get into bootloader on the affected side. The mod key method only works when the side is booting into ZMK

Joseph-Lion, 2024-07-13:
Unfortunately i've tried rotating the cable as well as different cables but nothing mounts.

KinesisCorporation, 2024-07-13:
Sorry to hear that!

It's possible there is a hardware failure on the left side. Please submit a trouble ticket and we'll issue an RMA for a repair/replacement assuming we can't solve the problem remotely.

https://kinesis-ergo.com/support/contact-a-technician/

closed 2024-07-13 by KinesisCorporation