KinesisCorporation/Adv360-Pro-ZMK #522: Cannot map calc hotkey as in Advantage 2 under KP+F7

AntoniMarcinek, 2024-07-17:
Hello,

I am trying to map Calculator hotkey to one of extra keys in Advantage 360 Pro to get the same behaviour as with Advantage 2 when pressing KP+F7. I tried both `C_AL_CALCULATOR` and `K_CALCULATOR` bindings, but none of these emit what KP+F7 emits. What can I do? Is there a way to check what numerical code is Advantage 2 emitting with KP+F7?

Best regards,
Antoni

ReFil, 2024-07-17:
A tool such as hid-recorder can tell you the exact key codes. Have you tried enabling nkro extended reporting in adv360_left_defconfig?

AntoniMarcinek, 2024-07-17:
I will try the hid-recorder. Now I tested with `CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT=y`, but it doesn't help. I do not know if it matters, but I am adding this to the `fn` layer. Maybe I should also try the base layer?

ReFil, 2024-07-17:
The layer doesn't matter at all

KinesisCorporation, 2024-07-17:
FWIW its probably worthwhile to try putting it in the top layer initially just to rule out a possible issue with accessing the other layer

AntoniMarcinek, 2024-07-18:
The problem was `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC=y`. Switched to `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y` and `&kp C_AL_CALCULATOR` started to work as expected. Thank you for pointing me to the NKRO, that allowed me to find the problematic option.

KinesisCorporation, 2024-07-18:
You seem to be combining syntax from the ZMK Pro version and the SmartSet USB version.

All of this is SmartSet only...
<base>
[tab]>[app]
[hk1]>[cxxs]
[hk3]>[keyt]
[esc]>[lshf]
[hk2]>[lctr]
[hk4]>[rctr]
[lctr]>[lmou]
[lalt]>[rmou]
[rwin]>[rshf]
[del]>[tab]
[home]>[esc]
[pgup]>[rwin]
[end]>[F4]
[pgdn]>[ralt]
<keypad>
&kp LC(C)
&kp LC(V)
&kp LC(X) 
[hk3]>[deft]
<function1>
[lshf]>[defs]
<function2>
<function3>

AntoniMarcinek, 2024-07-18:
@KinesisCorporation I don't understand what do you mean. Where did I combine syntax from the 2 versions?

Anyway I reopened, because `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y` enables `&kp C_AL_CALCULATOR` only through USB. While using Bluetooth, it doesn't work. Could it be that it is somehow associated with `CONFIG_ZMK_BLE_CONSUMER_REPORT_QUEUE_SIZE` (just a guess based on the fact that this option is for Bluetooth and that it says something about CONSUMER_REPORT).

KinesisCorporation, 2024-07-18:
Whoops, got two Github notifications at once. Sorry!

KinesisCorporation, 2024-07-18:
> @KinesisCorporation I don't understand what do you mean. Where did I combine syntax from the 2 versions?
>
> Anyway I reopened, because `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y` enables `&kp [...]

That's a good question for @ReFil

AntoniMarcinek, 2024-07-19:
@ReFil Thank you! I realised that I didn't set the `CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_FULL=y` in the right defconfig. After doing this everything started to work even without re-pairing. I did re-pair anyway just in case. Is there any info stating which options should be put in both defconfigs and which can go just to the left one?

closed 2024-07-19 by AntoniMarcinek