KinesisCorporation/Adv360-Pro-ZMK #523: help with defining 'macro type actions?' like cntrl+c, x, y etc on fn1 layer

mudduck, 2024-07-18:
refer to subject but im looking to on fn1 layer be able to press just the c key and have it run 'cntrl+c' etc tho am having trouble. I reached out earlier and got a prompt response from support for: 

```
&kp LC(C)
&kp LC(V)
&kp LC(X) 
```

but to no avail, pls audit and suggest changes to maybe work as expected if thats possible. here is full config for fn1 for ref: 

```
<base>
[tab]>[app]
[hk1]>[cxxs]
[hk3]>[keyt]
[esc]>[lshf]
[hk2]>[lctr]
[hk4]>[rctr]
[lctr]>[lmou]
[lalt]>[rmou]
[rwin]>[rshf]
[del]>[tab]
[home]>[esc]
[pgup]>[rwin]
[end]>[F4]
[pgdn]>[ralt]
<keypad>
&kp LC(C)
&kp LC(V)
&kp LC(X) 
[hk3]>[deft]
<function1>
[lshf]>[defs]
<function2>
<function3>
```

KinesisCorporation, 2024-07-18:
Sorry for the confusion. This is a forum for the 360 Pro Bluetooth with ZMK but it looks like you actually have a SmartSet 360 which has a totally different programming engine. The original instructions were for a Pro so please disregard. 

For SmartSet here are the 3 basic macros you want to add under the <Function 1> header in the desired layout files. 
{c}>{-lctr}{c}{+lctr}
{v}>{-lctr}{v}{+lctr}
{x}>{-lctr}{x}{+lctr}

These would be accessed by pressing AND HOLDING the "fn" pinky keys and the tapping C, V or X respectively.

mudduck, 2024-07-21:
appreciate your response out of band, much appreciated. thx.

closed 2024-07-21 by mudduck