KinesisCorporation/Adv360-Pro-ZMK #527: Build Artifacts Missing in firmware Directory After Running make all Using Podman

LandonSchropp, 2024-07-21:
I have Podman installed on my machine and I'm encountering an issue where no build artifacts are generated in the firmware directory after executing the `make all` command.

### Repro

```
git clone https://github.com/KinesisCorporation/Adv360-Pro-ZMK.git
cd Adv360-Pro-ZMK
make all
```

I'm attaching the [output](https://github.com/user-attachments/files/16325353/output.txt) from `make all` as a text file since it's too long for GitHub's character limit.

### Environment

Operating System: macOS 14.5
Podman Version: 5.1.2
Make Version: 3.8.1

LandonSchropp, 2024-07-21:
I tried it with `Docker` as well and I'm also not seeing any output there.

mattrayner, 2024-07-21:
> I also tried it with Docker and I'm not seeing any output.

Hi Landon, have you tried pushing to GitHub and looking at the GitHub
Actions Artifacts?

Usually you’d get the output from there

Matt Rayner
http://www.mattrayner.co.uk/

LandonSchropp, 2024-07-21:
Ah, never mind. The files are there after all. I didn't realize that `exa` hides files that are ignored by Git.

LandonSchropp, 2024-07-21:
Thanks for the help @mattrayner!

closed 2024-07-21 by LandonSchropp