KinesisCorporation/Adv360-Pro-ZMK #536: Windows 10 Bluetooth randomly disconnects and reconnects

jmarais, 2024-08-07:
# Problem
The Bluetooth will randomly disconnect and reconnect after about 5-10 seconds.
# Attempted Solutions
I looked through all the past issues regarding Bluetooth for info. 
I tried the `settings-reset-v3.0` on the left/right halves and then formatting the latest V3 (with my key profile) and `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=n` in the left/right config files.
# Other info
For reference here is my diff against the V3 branch:
[diff against V3](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/compare/V3.0...jmarais:Adv360-Pro-ZMK:V3.0.1)

I am connecting to my motherboard's Bluetooth with this network adapter (about 1meter away):
[Intel Dual Band Wireless-AC 3168 ](https://www.intel.com/content/www/us/en/products/sku/94854/intel-dual-band-wirelessac-3168/specifications.html)
With firmware version: `HCI 8.4352 / LMP 8.4352`

The DC happens quite frequently to the point that I am creating this issue asking for anything suggestions I might have missed.

ReFil, 2024-08-08:
Intel Wireless can be problematic, although the experimental features flag should make things work

jmarais, 2024-08-08:
I will switch it back OFF again and see what happens.

jmarais, 2024-08-08:
So steps I took.
1. Switch off both sides. 
2. Remove the adv360 device from my windows bluetooth devices.
3. flash left then right with the `settings-reset` firmware.
4. downloaded my firmware with [CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=n](https://github.com/jmarais/Adv360-Pro-ZMK/actions/runs/10301243527)
5. flashed left and then right with the new firmware (about 30minutes ago).
6. pair the device as a new device in windows.

It just did a disconnect/reconnect loop.

jmarais, 2024-08-12:
If there is no other config options or values to try, is it possible to recommend a USB Bluetooth receiver model/brand/vendor that is known to work with the Bluetooth module in the keyboard?

ReFil, 2024-08-21:
We've had good success with the TP link UB4A

jmarais, 2024-08-21:
Thanks. I will try that receiver.

closed 2024-08-21 by jmarais

jmarais, 2024-08-28:
Been using the tp-link UB5A for about 5 days now. No issues to report. Thanks.

KinesisCorporation, 2024-08-28:
Great. Please keep us posted.