KinesisCorporation/Adv360-Pro-ZMK #544: Kinesis Form Split Touchpad Keyboard: Function Key Behavior and External Monitor Brightness Control Issues on MacBook Pro

mkwng, 2024-08-21:
I'm experiencing two issues with my Kinesis Form Split Touchpad keyboard on my MacBook Pro:

1. Function Key Behavior:
   - The keyboard doesn't respect macOS's function key behavior settings.
   - Expected behavior: Pressing F9 and F10 keys should adjust brightness without using the Fn modifier key, as per macOS settings.
   - Actual behavior: Fn modifier key is required to use F9 and F10 for brightness adjustment.
   - macOS setting "Use F1, F2, etc. keys as standard function keys" is turned OFF, which should allow direct brightness control.

2. Brightness Control:
   - The brightness adjustment keys only affect my MacBook Pro's screen brightness.
   - They do not adjust the brightness of my external monitor, regardless of which display is currently active.
   - Expected behavior: Brightness controls should adjust the active display, including the external monitor.

System Information:
- Keyboard: Kinesis Form Split Touchpad
- Computer: MacBook Pro 14-inch (November 2023 model)
- macOS Version: Sonoma 14.6.1 (23G93)
- Function Key Setting: "Use F1, F2, etc. keys as standard function keys" is OFF (see attached screenshot)

Steps to Reproduce:
1. Connect Kinesis Form Split Touchpad keyboard to MacBook Pro
2. Ensure "Use F1, F2, etc. keys as standard function keys" is OFF in System Settings
3. Press F9 or F10 key (without Fn modifier) - brightness does not change
4. Press Fn + F9 or Fn + F10 - brightness changes on laptop screen only, not on external monitor

Expected Behavior:
- F9 and F10 keys should adjust brightness without needing the Fn modifier
- Brightness controls should affect the active display, including external monitors

Actual Behavior:
- Fn modifier is required to use F9/F10 for brightness adjustment
- Brightness adjustment only affects the laptop screen, not the external monitor

Any assistance in resolving these issues would be greatly appreciated.

ReFil, 2024-08-21:
Hi, this repository is only for the Advantage 360 Pro only, not the Form. Furthermore the setting you mention is most likely exclusively for Apple branded keyboards and thus the form will not respect it. The Form's Fn key is not equivalent to the apple Fn key

KinesisCorporation, 2024-08-21:
Hi @mkwng

closed 2025-01-16 by KinesisCorporation