KinesisCorporation/Adv360-Pro-ZMK #555: Unable to pair - Windows 10 Enterprise 22H2 - Bluetooth encryption key too small

MasaiMarmot, 2024-09-09:
Keyboard pairs instantly with my iPhone - no issues.  However, I've had zero luck connecting it with my PC. 

Following standard practice for pairing a Bluetooth device to my work machine (HP Elitebook 840 G10 running Windows 10 Enterprise 22H2). ADV360 initially looks like it's paired (Windows says, "Success! Your device is ready for use", but the ADV360 Profile light continues to flash rapidly, and the status of the paired device when viewed in Bluetooth settings oscilates between "Paired" and "Connected" constantly. Keyboard sends no input to the machine.

I've done multiple reboots of both keyboard and laptop. I've installed 4 different iterations of the firmware - no change.

Found the following message in the System logs in event viewer:
Source: BTHUSB
Event ID: 49
General: Windows rejected a connection from your Bluetooth device ((f5:9f:f4:62:72:37)) because the resulting encryption key size was smaller than the system required minimum.

Following suggested guidance from the Kinesis support crew, I followed the guidance from petejohanson in [this thread](https://www.reddit.com/r/zmk/comments/10f8jdn/comment/j6jpu06/).
I modified the setting of CONFIG_ZMK_BLE_PASSKEY_ENTRY from "n" to "y" in adv360_left_defconfig, rebuilt my gitHub branch, and installed the new firmware.  Now my laptop prompts for a 6 digit passcode, but the pairing process after successful passcode entry is exactly as described above - no change.

I'd love to keep this keyboard, but pairing with my work PC is a must. BTW - tried using a BT dongle for pairing as well, same result.

ReFil, 2024-09-10:
This sounds like either the host or the 360 has old pairing keys. Go through each profile on the 360 pro and unpair it explicitly (fn + win on the default keymap). Ensure it's unpaired on the pc too using "remove device" and try pairing again. I'd recommend removing the passkey entry config option too.

If one of the Bluetooth profiles on the 360 Pro hold the same Mac address as a host you're trying to connect on another profile it will usually fail.

MasaiMarmot, 2024-09-10:
Thanks for the reply. I went through those steps multiple times before I started digging into Event Viewer to try to figure out what was going on.  Just did it again to triple/quadruple check. 

1. Wiped each BT setting on the keyboard (Mod+WIN) for each bay (Mod 1, Mod 2, etc.)
2. Removed Adv360 Pro from BT settings on laptop
3. Opened device manager, clicked "View", clicked "Show hidden devices"
4. "Deleted Adv360 pro rt" from the "Other Devices" section of the device manager
5. Removed "Adv360 pro" from the "Bluetooth" section of device manager
6. rebuilt branch 3.0 of the firmware, reverted the passkey entry config from y to n
7. flashed the firmware on the adv360 pro. 
8. powercycled the keyboard 3 times.

Exact same behavior, same error message generated in Event Viewer.

"Windows rejected a connection from your Bluetooth device ((f5:9f:f4:62:72:37)) because the resulting encryption key size was smaller than the system required minimum."

What else could I try? at this point the keyboard is unusable. I already have 2 corded versions of the Advantage - the only reason to by the 360 pro was to be wireless.
Thanks

ReFil, 2024-09-10:
Have you tried the settings reset procedure?

ReFil, 2024-11-27:
Hi, just following up on this, were you able to try the settings reset procedure? Did it make any difference?

ReFil, 2024-12-18:
That's very strange, maybe your work have some group policy requirements set up to mandate very fancy encryption. As far as I know ZMK/Zephyr complies with all applicable standards for ble encryption. The only reference I can find to the encryption key too small error is for Bluetooth classic, not Bluetooth low energy. Have you tried a different Bluetooth adapter on your work PC?

MasaiMarmot, 2024-12-19:
Yeah - I even went so far as to buy a bluetooth dongle. Same result.