KinesisCorporation/Adv360-Pro-ZMK #560: LEDs burning my eyes.

adofse, 2024-09-17:
As mentioned https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/129,  LEDs brightness intensity is too strong (100%?) which results in burning my eyes.

## To reproduce
I observed the following two patterns that LEDs shine 100%.
- a. Turn on power
- b. Double click the Bootloader Button (to flash firmware)

## Expected Behaviour
User can change LED brightness intensity.

ReFil, 2024-09-17:
The short bright flash when the board powers up andd is in bootloader is unavoidable I'm afraid, it is an artifact of the bootloader and cannot be configured

adofse, 2024-09-17:
@ReFil Does that mean you can't change that behavior in https://github.com/ReFil/zmk/tree/adv360-z3.5 or any other dependencies ?

Seeing too bright lights could cause serious health harms for human eyes so it should be cared by Kinesis Corporation responsibility.

adofse, 2024-09-20:
Now we know how @KinesisCorporation treats their users since it's been 3 days when I report potential health harms issue.

I believe it's too early to take advantage open source ZMK for @KinesisCorporation  as they claim they area small private company. 

Highly Disappointed!

KinesisCorporation, 2024-09-20:
Unfortunately this is not user-configurable. 

Please contact Tech Support and link them to this discussion. We will accept a return of the keyboard.

https://kinesis-ergo.com/support/contact-a-technician/

adofse, 2024-09-27:
@KinesisCorporation 

> Unfortunately this is not user-configurable.

Please elaborate this. Editing source code in https://github.com/ReFil/zmk/tree/adv360-z3.5 could solve this problem?
If not, are you willing to contact someone who develop bootloader firmware?

KinesisCorporation, 2024-09-30:
Hi,

The next time we do work on the bootloader we will explore whether the LEDs can be dimmed. If you decide to keep the keyboard we suggest not looking directly at the LEDs while opening the bootloader. We are still happy to process a return/exchange of the keyboard.

closed 2024-09-30 by KinesisCorporation

adofse, 2024-10-02:
> The next time we do work on the bootloader we will explore whether the LEDs can be dimmed.

@KinesisCorporation Could you assign @ReFil to take a look?