KinesisCorporation/Adv360-Pro-ZMK #571: Advantage2 writing gibberish

telltellcopywriting, 2024-10-09:
Hi,

I'm having issues with my Advantage2 keyboard. It's plugged into a Mac and I'm using Monterray. It was working fine until 2 days ago, when it started being really buggy and typing erratically. Here's a quick video. Ocassionally the keypad gets triggered. I tried a soft and hard reset, I attempted to reprogram the keys, I turned everything off and on, I cleaned the keyboard and it still doesn't work. I've also reprogrammed it for Mac and ensured it's on qwerty and not dvork.

KinesisCorporation, 2024-10-09:
Hi, @telltellcopywriting 

Sorry to hear you are having problems with your Advantage2! Its possible you accidentally recorded one or more Macros so a Hard Reset should take care of that. If that doesn't work, please submit a trouble ticket here and we can help you troubleshoot and determine your warranty status in the event a repair is required.

https://kinesis-ergo.com/support/contact-a-technician/

This forum is for Advantage360 discussions only.

closed 2024-10-09 by KinesisCorporation

telltellcopywriting, 2024-10-17:
Hi,

I’ve submitted 3+ support tickets but have not heard back.

https://kinesis-ergo.com/support/contact-a-technician/