KinesisCorporation/Adv360-Pro-ZMK #589: Help--no idea what I'm doing

lpbowlingdvm03, 2024-11-03:
Hello, 

my primary reason for getting this keyboard was for arthritis in my hands and elbows.  I use a Mac.  I have managed to figure out how to do the quick files to change the keys for the thumbs to a mac format but I would love code to configure the whole keyboard if someone has already done that.  I have never coded in my life and have tried to read how to do it but it's just not making sense to me.  Any help is appreciated.  

I do plan on learning code in the future but I also have a full time job hence this keyboard that helps my hands but is breaking my brain!

closed 2024-11-03 by lpbowlingdvm03

Omniscently, 2024-11-19:
Hey there, submit a ticket here: https://kinesis-ergo.com/support/contact-a-technician/

I will get you sorted. :)

lpbowlingdvm03, 2024-11-19:
Hi Ty,

Thank you so much for responding!  I managed to figure out what I needed to
after watching a few videos and rereading the manual.

I thought I closed that thread (which is more proof that I have no idea
what I’m doing!).

Lauren