KinesisCorporation/Adv360-Pro-ZMK #601: Advantage 360

Bimka123, 2024-11-23:
I have an Advantage 360 UBS. The I press the Pagu Up key, the Windows start menu is displayed. When I press the Page Down key, the page is highlighted and futher taps on the key have no effect. I have tried reprogramming the keys, but of course the functions that I select to replace them are still the Page Up and Page Down functions, which make exactly the same mistakes.

KinesisCorporation, 2024-11-23:
That sounds like you have some accidental programming on those keys. Try doing a Soft Reset described on page 18 of the manual linked below. 

https://kinesis-ergo.com/wp-content/uploads/Advantage360-SmartSet-KB360-Users-Manual-v10-12-22.pdf

This forum is for Advantage360 Pro which has different electronics so if that doesn’t work please submit a trouble ticket.

closed 2024-11-23 by KinesisCorporation

Bimka123, 2024-11-24:
That worked, thanks!

https://kinesis-ergo.com/wp-content/uploads/Advantage360-SmartSet-KB360-Users-Manual-v10-12-22.pdf