KinesisCorporation/Adv360-Pro-ZMK #604: [Warning!] Kinesis doesn't take seriously about burning users eyes!

adofse, 2024-11-28:
I opened the following the two issues but Kinesis Corporation was reluctant to state the official statement and close those issues without responding for like one month!
- https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/560
- https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/570

Those issues are related to harming users eyes so it should not be ignored!
At least provide WHY you can't fix it or TIMELINE to fit it.

So disappointed how Kinesis Corporation treats precious user's feedback.
They want fix this complaint by just RMA😂 It's easy for them by just not fixing it. That's a shame.

danielleevandenbosch, 2025-01-18:
What are you referring to?

aaronjensen, 2025-06-08:
@adofse Have you tried adjusting `CONFIG_ZMK_RGB_UNDERGLOW_BRT_MAX` and/or `CONFIG_ZMK_RGB_UNDERGLOW_BRT_START` to something less than `100`? I haven't tried it, just came across it.