KinesisCorporation/Adv360-Pro-ZMK #611: Assign a keybinding to ctrl+alt

dusty-phillips, 2024-12-21:
I'd like the (2) key to perform the same thing as holding down ctrl+alt at the same time. So e.g. 2+a would invoke `ctrl+alt+a` and 2+v would invoke `ctrl+alt+v`.

I thought `&kp LCTRL(LALT)` would get me there, but I get errors (e.g. https://github.com/dusty-phillips/Adv360-Pro-ZMK/actions/runs/12448278325/job/34752691663). I've tried various things with macros and mod-morph, and have caused chatbots to break out in tears. Is there a secret to this behaviour?

ReFil, 2024-12-21:
Have you tried `&kp LC(LALT)` ? I'm not sure LCTRL() is a valid key modifier

dusty-phillips, 2024-12-21:
I had not tried that. I had tried `LC(LA)`, which fails to build. `LC(LALT)` builds, but when I install the firmware, it doesn't register the `ctrl+alt+...` keystroke. 😞

closed 2024-12-21 by dusty-phillips