KinesisCorporation/Adv360-Pro-ZMK #618: Bluetooth Connection

kathyO42, 2024-12-31:
Windows 10 operating system.  Advantage Pro Keyboard.  Will not connect via bluetooth.  Keyboard works fine with USB cables.  I have updated the firmware (Jan 2024).  Still not able to connect via bluetooth.  No issue with laptop as I am able to connect other devices via bluetooth.   Keyboard will connect with Iphone easily, but not laptop.   I work for a bank and have connected with their IT team and they have indicated that there are no blocks for connecting via bluetooth.  Need assistance or I will be forced to return the keyboard, which would be a shame.

KinesisCorporation, 2024-12-31:
Hi Kathy,

Sorry to hear you are having problems pairing with your laptop. Have you selected a different Profile on the keyboard from the one that has been successfully paired with your phone?

Please send us a trouble ticket if that doesn’t work and we’ll have you get it working. 

https://kinesis-ergo.com/support/contact-a-technician/

closed 2024-12-31 by KinesisCorporation

kathyO42, 2024-12-31:
I have submitted a ticket, with no response as of yet.  Ticket is E-27963.  I removed the device from my Iphone and honestly just paired it to the Iphone to help eliminate possible issues.   I have tried all profiles and keep getting the same error which is “Try connecting your device again”.   The keyboard is throwing the signal but the connection never works.  This is a refurbished keyboard purchased directly from Kinesis and I am wondering if that is the problem.

KinesisCorporation, 2024-12-31:
If it’s connecting to one device successfully it’s unlikely a hardware issue.

Is the Profile LED flashing fast? If it’s not you’ll need to do Mod+Windows to clear the profile on the keyboard first. 

Can you clear/forget the keyboard from your BT menu?

We’re closed until Thursday for the holidays but will do our best to get back to you by Friday.

kathyO42, 2024-12-31:
That worked!  Thank you!One other question - Is there a way to reduce the sensitivity of the keys?  I am getting random keystrokes often.