KinesisCorporation/Adv360-Pro-ZMK #622: Mouse movement

Gazareth, 2025-01-05:
Last month zmk merged mouse movement capabilities.

https://github.com/zmkfirmware/zmk/pull/2477

Just wondering if/when this is gonna make it to Adv360?

grantslape, 2025-01-14:
I tried naively adding `CONFIG_ZMK_POINTING=y` to `adv360_right_defconfig` and `adv360_left_defconfig` and `make` blew up during `west build`:

```
/app/config/boards/arm/adv360/adv360_left_defconfig:77: warning: attempt to assign the value 'y' to the undefined symbol ZMK_POINTING
Parsing /app/zmk/app/Kconfig
Loaded configuration '/app/config/boards/arm/adv360/adv360_left_defconfig'
Merged configuration '/app/zmk/app/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /app/zephyr/cmake/modules/kconfig.cmake:355 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /app/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /app/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /app/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/app/build/left -GNinja -DBOARD=adv360_left -DZMK_CONFIG=/app/config -S/app/zmk/app
make: *** [all] Error 1
```

ReFil, 2025-01-17:
Hi everyone, we have been working on pointing support in the 360 pro and this is now in an open PR for testing prior to merging: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/pull/630

ReFil, 2025-02-06:
Closing this issue now as mouse movement and click support is in V3.0 with #630