KinesisCorporation/Adv360-Pro-ZMK #626: Keyboard Won't Connect via Bluetooth

rjparton, 2025-01-14:
- Bluetooth won't connect to my Mac.
- It works when I connect via cable. 
- Please review the screenshots attached. 
- It shows in "Nearby Devices". 
- When I click connect, I get the spin wheel for 1 second, then "Disconnect" for a split second, then it returns to "Connect". 
- It does not leave the "Nearby Devices" section. I can repeat this cycle indefinitely
- I followed all the instructions provided over email and got to the "Very Very Hard" stage which is submitting this ticket

[image]
[image]

ReFil, 2025-01-14:
Is the bluetooth LED rapidly flashing? has the board been paired to any other devices?

rjparton, 2025-01-14:
Hi there, I just managed to fix it. The instructions referred to the Window key but I remapped this. I looked at an image to see where it originally was and it worked. Thank you!

closed 2025-01-14 by rjparton