KinesisCorporation/Adv360-Pro-ZMK #636: Issue building my keyboard keymap

benileo, 2025-01-31:
[Build](https://github.com/benileo/Adv360-Pro-ZMK/actions/runs/13079298421/job/36509026841#step:1:25)
This request has been automatically failed because it uses a deprecated version of `actions/upload-artifact: v3`. Learn more: https://github.blog/changelog/2024-04-16-deprecation-notice-v3-of-the-artifact-actions/

Is there an update needed in order to fix this? When I try to "sync" with source repo I get a bunch of merge conflicts.
https://github.com/benileo/Adv360-Pro-ZMK/actions/runs/13079298421

ReFil, 2025-02-03:
What files conflict? If you don't want to sync fork you could just update the upload artifact in build.yml from V3 to v4

benileo, 2025-02-03:
There are two issues. The build is broken on V3 for my fork. The second issue is I can't sync the V3 fork due to conflicts. 

What do you recommend? Should I try and upgrade to V4? Will that fix the issue of the deprecated upload artifacts that your CI job is trying to use?

ReFil, 2025-02-04:
If you change @v3 in this line to @v4 https://github.com/benileo/Adv360-Pro-ZMK/blob/V3.0/.github%2Fworkflows%2Fbuild.yml#L56

Your builds should succeed again. 

Alternatively you can go through the git conflict resolution process to get the latest upstream code from this repo, which was updated  here https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/b4165d65c86709050e37cc98916f687659422605 to fix the deprecation issue you're encountering

ReFil, 2025-02-06:
Did updating your repo solve the problem?

ReFil, 2025-02-12:
Looking at your actions it seems like the issue has been solved so i'm closing this issue as solved

closed 2025-02-12 by ReFil