KinesisCorporation/Adv360-Pro-ZMK #642: Failing to build

dylanmmckay, 2025-02-07:
Hi! I apologize that I don't know the proper ettiquette here. I just got my keyboard, forked the repository, and I haven't been able to successfully complete a build. Here is a link to the build action: https://github.com/dylanmmckay/Adv360-Pro-ZMK/actions/runs/13192343327/job/36827699730

If there is anything I can provide to be helpful, please let me know!

KinesisCorporation, 2025-02-07:
Hi Dylan,

We're in the process of migrating to a new programming paradigm and away from GitHub. 
The new recommended tool is called Clique and it bypasses GitHub completely: https://clique.kinesis-ergo.com/ (Use Chrome or Edge Browser).

If you prefer to use Github we recommend configuring with this editor: https://nickcoutsos.github.io/keymap-editor/.

dylanmmckay, 2025-02-09:
I have been using the editor you're recommending, but I think the build issues are a problem with the code that is currently on GitHub.  Someone on Reddit [sent me a fix in a line of code](https://www.reddit.com/r/kinesisadvantage/comments/1ijsast/comment/mbhx0ra/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) which helped me get some successful builds. I just tried to add a macro press to my layout, and the builds are failing again: https://github.com/dylanmmckay/Adv360-Pro-ZMK/actions/runs/13227870576

ReFil, 2025-02-09:
Please try syncing your repository with the latest upstream using the "sync changes" button. Please note this may cause a git conflict with the macro you just added

dylanmmckay, 2025-02-10:
Thanks! Both issues seem to be fixed.

closed 2025-02-10 by dylanmmckay