KinesisCorporation/Adv360-Pro-ZMK #682: Alt-Key doesn't work in V3.0 build?

davidyang, 2025-03-09:
Hey all - I'm flashing the left build to an Advantage 360 ZMK and then using Keyboard Events Viewer I can no longer activate the Alt key.  What's weird is that Karabiner EventViewer can see the alt, but the Keyboard event viewer doesnt.  Any ideas what's going on?  I've tried disabling Karabiner without any effect.

I have to go back to commit 8988c99a to get Alt key working again.

ReFil, 2025-04-16:
Have you tried unpaired and repairing the connection to the host pc?

ReFil, 2025-05-22:
Just following up, are you still having this issue?

davidyang, 2025-05-22:
Thanks for following up - it was a different issue being caused by
Karabiner.

ReFil, 2025-05-22:
Ok awesome, i'm going to close this issue out as solved in that case. Thank you for your prompt reply

closed 2025-05-22 by ReFil