KinesisCorporation/Adv360-Pro-ZMK #699: Bluetooth profile stops functioning

tbenst, 2025-04-15:
Continuation of https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/282.

If my laptop goes to cusp of bluetooth range for ~30 minutes, the bluetooth profile stops functioning until reset. 

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/282

1. using keyboard with laptop on desk
2. close lid, laptop moved to another room (~40ft w/ one wall), keyboard left on table (switches in on position)
3. Use laptop for 30 min (possibly less)
4. close lid, move laptop next to keyboard on desk
5. Notice that keyboard not paired, power toggling does not fix

Unfortunately this now happens for me about 1-2x per day, whenever I take my laptop to a particular conference room for a meeting while leaving my keyboard on my desk.

Mitigations:
- can reset bluetooth profile and repair
- can prevent issue from occurring by remembering to turn off keyboard before moving laptop to the other room ~40 feet away
- issue does not occur if move laptop well outside of bluetooth range (eg 1 mile; presumably less too) while keyboard is left on

KinesisCorporation, 2025-04-15:
Sorry to hear this! A lot of these connectivity issues were addressed in the February 2025 firmware update. Are you running the latest version? 

If you are and this is still happening we can investigate an upstream ZMK-level fix.

ReFil, 2025-04-16:
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config%2Fboards%2Farm%2Fadv360%2Fadv360_left_defconfig#L50 if you comment out (by prefixing with a #) or remove this line it will decrease the transmit power of the left side. This might help your situation by allowing the board to fully disconnect at the edge of the range

tbenst, 2025-04-21:
Thanks! I’ll first try the Feb 2025 firmware, and if that doesn’t resolve the tip from ReFil. Will report back (or future readers feel free to ping me if I haven’t!)

ReFil, 2025-05-22:
Hi @tbenst did the updated firmware sort your issue out?

closed 2025-11-06 by ReFil