KinesisCorporation/Adv360-Pro-ZMK #723: Default control of RGB lights

aaronjensen, 2025-06-07:
Hi,

I'm wondering if it's possible to disable the profile light and the layer light. I'd like to leave the RGB lights on other than those (caps lock, battery check, etc.) Is that something that can be done with the config here, or does that require modifying the zmk fork?

Thanks!

KinesisCorporation, 2025-06-07:
Yes!

Mod - Space is toggles all indicators on / off in the default layout. 

It should also be available in Clique.

aaronjensen, 2025-06-07:
Sorry, that's not what I meant. I meant I want it that way when the keyboard is plugged in. I also don't want *all* lights off. I want the caps lock light to still be functional.

KinesisCorporation, 2025-06-07:
It’s all or nothing.

aaronjensen, 2025-06-07:
> @ github. com>

Indeed, which is why I'm asking if there's a way to do it even if it requires zmk modifications. I'm also asking about the power on (default behavior). Even if I could have the default state of all lights be off, that would be useful. Every time I plug my dock in I have to turn the lights off on the keyboard. It's annoying to have to do every time.

Aaron

ReFil, 2025-06-08:
The default power on state can be changed by changing this line https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config%2Fboards%2Farm%2Fadv360%2Fadv360_left_defconfig#L38 to =n on both sides. Currently there is no way to control the lights individually. That would need to be changed in the actual zmk firmware itself rather than the config repo. If you fork the fork the 360 pro uses you can point your config repo to that fork and branch and make changes. The file you'll want to change is "rgb_underglow.c"

aaronjensen, 2025-06-08:
@ReFil Unfortunately, I can't seem to get the changes in my fork to do anything. 

I've updated west.yml: https://github.com/aaronjensen/Adv360-Pro-ZMK/commit/6579486f2f67e88ef41def99a2286257809b3f46

And here's an example change:  https://github.com/aaronjensen/zmk/commit/e6a3f623bdd5a2c981035d9788053f2cd2c9ed57

I ran `make clean_image` and then `make`. After installing the new firmware, I get the exact same LED behavior. Do I need to build zmk in the zmk repo first or anything like that?

aaronjensen, 2025-06-08:
Nevermind. I figured it out. It was docker caching. I had to update the revision in west.yml each time I wanted a different build (or completely purge docker cache)

closed 2025-06-08 by aaronjensen